cifertech/nRFBox #6: pcb

Clayrecovery20, 2023-02-27:
can i get pcb design?

jlescalonap, 2024-08-23:
@cifertech could you sell the PCB design?

cifertech, 2024-09-04:
I have uploaded the PCB files for nRF-BOX V2.

https://github.com/cifertech/nRFBox/tree/main/PCB

zRCrackiiN, 2024-09-20:
I've assembled 2 now, both seem to have power issues, LF33 gets hot real fast I'm not even sure where to start diagnosing it haha

cifertech, 2024-09-20:
Hi @zRCrackiiN 
Please provide photos of your assembled PCB and test your nRF-BOX without the C8 and C9 capacitors.

jlescalonap, 2024-09-20:
Sorry if this is not the right place to answer but, where did you order the PCBs? @zRCrackiiN

zRCrackiiN, 2024-09-20:
> Sorry if this is not the right place to answer but, where did you order the PCBs? @zRCrackiiN

pcbway

cifertech, 2024-09-20:
> im not sure i understand the power switch, when switched down (linking bat to BT1) it links the battery to the 4056, but when its up it links BT1 to +5v?

It's a clumsy logic design for controlling battery charging and powering on/off when the battery is connected, but it works.

If you are still experiencing issues with LF33, try to locate a short circuit. It's better to do the first tests with just one nRF24.

By the way, it's nice to see the nRF-BOX in purple!

zRCrackiiN, 2024-09-20:
I have a second board without the esp and the nrf modules, it's doing the same thing, definitely down the bottom of the board somewhere

cifertech, 2024-09-20:
> I have a second board without the esp and the nrf modules, it's doing the same thing, definitely down the bottom of the board somewhere

**Just following up on the LF33.**

[image]

[image]

zRCrackiiN, 2024-09-21:
> > I have a second board without the esp and the nrf modules, it's doing the same thing, definitely down the bottom of the board somewhere
> 
> **Just following up on the LF33.**
> 
> ![photo_5436097977669245949_y](https://private-user-images.githubusercontent.com/62047147/369573477-75bb72bb-770f-4cb3-b0a8-b9e827ba4fef.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4NzgzMjksIm5iZiI6MTcyNjg3ODAyOSwicGF0aCI6Ii82MjA0NzE0Ny8zNjk1NzM0NzctNzViYjcyYmItNzcwZi00Y2IzLWIwYTgtYjllODI3YmE0ZmVmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIxVDAwMjAyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1YTc1YTBmMzliZThlMjI5MDNiZGVmZjAxMjViYWFiNTQwYWI4NjUwYzBkMDk0MjIyNjExZWU4NTI5NzAzM2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.4zpdHV5s7yixaU8DHSbkNRZw5-Bn7qoqz3hx25CS5fs)
> 
> ![photo_5436097977669245950_y](https://private-user-images.githubusercontent.com/62047147/369573484-13f54f11-2be0-46a4-9408-967241700a38.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4NzgzMjksIm5iZiI6MTcyNjg3ODAyOSwicGF0aCI6Ii82MjA0NzE0Ny8zNjk1NzM0ODQtMTNmNTRmMTEtMmJlMC00NmE0LTk0MDgtOTY3MjQxNzAwYTM4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIxVDAwMjAyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM1NzgxNzNhYmUxNTFmNTIyZGNiZGQ1YTI2YmQxYmVmZGZkNjUyZTIyMTI4NTlkOWJiYjJlNDNhMWE5OTg0YWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.okaK3MBvRIb204g3rbQMYdcRYm1WwexB5THaHdTf6o4)

im currently looking at this, i found that it was going to ground when not powered on the 5v side, when powered it stops going to ground, and powers up, but there is no output.

im also noticing that mine say ST GN LF33, not like yours, i cant find a spec sheet for the GN variant so i might need to try source other LF33 chips

zRCrackiiN, 2024-09-21:
basically, ive removed the GN LF33, added bench power to the 3v3 circuit and plugged in the usb, i was able to program and test, working well! i wonder if it really is just that reg!

jlescalonap, 2024-09-24:
> basically, ive removed the GN LF33, added bench power to the 3v3 circuit and plugged in the usb, i was able to program and test, working well! i wonder if it really is just that reg!

Could you please share how it enden up? @zRCrackiiN

zRCrackiiN, 2024-09-24:
> > basically, ive removed the GN LF33, added bench power to the 3v3 circuit and plugged in the usb, i was able to program and test, working well! i wonder if it really is just that reg!
> 
> Could you please share how it enden up? @zRCrackiiN 
> 

I'll share after I've got the new lf33 added to the board, hopefully that fixes my issues

zRCrackiiN, 2024-11-20:
> Hey there @zRCrackiiN did you manage to solve the problem?

Yes, it appears to be bad LF33 components, my first order was from aliexpress, none of them worked.

Second order was from digikey and they work great

illlumin, 2024-12-06:
@zRCrackiiN,  could you share the 3D printing file? I’ve already ordered the circuit board and am currently looking for an enclosure for it. thank you

zRCrackiiN, 2024-12-30:
> @zRCrackiiN, could you share the 3D printing file? I’ve already ordered the circuit board and am currently looking for an enclosure for it. thank you

hey mate,
currently still in development but getting closer 
[image]

Jeremiah-oAT, 2025-01-18:
I'm looking to have PCBWay completely assemble a few of these units. Does anyone have the pick and place files?