cifertech/nRFBox #11: PCB Gerber File

jayrogogn, 2024-09-02:
Hi, I didn't find the Gerber file in the main menu.
It is not in the "PCB" folder.
Could you make it available?

cifertech, 2024-09-04:
Hi @jayrogogn 
I added the Gerber file, but you can also use .pcbdoc to order the PCB. Please ensure there are no issues before placing the order.

https://github.com/cifertech/nRFBox/tree/main/PCB

knlklabacka, 2024-09-09:
Good morning!  I uploaded the PCB file to JLCPCB and there are issues with the PCB that I'm not sure how to address.  Can you help?  The messages are from JLCPCB.

First issue:
As shown for 2 layer board ,the minimum via size is 0.3mm and the minimum via diameter is 0.5mm, can we increase the hole size and diameter to meet our capabilities?

Second issue:
slots in outline layer will cut the traces, should we delete them to proceed?
[image]

atsigr, 2024-10-12:
@cifertech i guess the pcb is with altium? Just a question, can you share the altium project so maybe we can adjust the pcb maybe for some of the materials ?