cifertech/nRFBox #13: spi communication problem

CtlAltDel6, 2024-09-14:
Breadboard build only works with a single radio attatched to the spi channel. Built with nodemcu wroom32 and using nrf24l01 + modules and breakout boards. I have only a single channel scope so I am unable to watch both ce and csn lines to confirm signal timing and transmition from multiple radios concurrently. Have confirmed spi coms as the issue by running diodes to spi on second and third radios. With that setup I get three radios active in multi jaming and a reliable signal in scanner and analyzer feom radio 1, but of course no mesurable signal from radio 2 or 3.
I have tried 2 different proto board variations hoping that my jumper wires were at fault but both have failed.

opskrip, 2024-09-16:
Hello. I have the same story, but I don't understand it. Can you post a photo of how you did it?

CtlAltDel6, 2024-09-16:
[image]
I connected radio 1 straight to the spi bus pins and used three high speed switching diodes to connect both radio 2 and 3 together to the spi pins. This allows for the prepare radio logic to run and returns active on multi ble jammer.
[image]

[image]

CtlAltDel6, 2024-09-16:
[image]
Both prototype builds have the same issue

CtlAltDel6, 2024-09-16:
[image]
Both prototype builds fail in the same way

opskrip, 2024-09-17:
Thank you very much. I'm new to arduino, so I'm learning how to code through third-party projects. Have a nice day

cifertech, 2024-09-20:
Hi @CtlAltDel6 
as I tested the project with 3 nRF24, your problem might be a power issue, but the SPI communication is always on the table, I will try to test to see if I can find the problem. If both prototypes fail in the same way, there might be something wrong with my code, but I will try to test it again.

By the way, that's a really impressive prototype.

CtlAltDel6, 2024-12-29:
[image]
[image]
Fixed in last update, rushed a messy case. 8.2" long, 3.6" wide, 1.2" depth including antenna. Thank you for all your hard work Cifertech

CtlAltDel6, 2024-12-29:
Closed, fully functional coms after update

closed 2024-12-29 by CtlAltDel6