cifertech/nRFBox #16: OLED screen not working

logm1lo, 2024-09-28:
When I complied and flashed the code to esp32 WROOM U and connected the OLED screen to the esp32. The screen won't boot

logm1lo, 2024-09-28:
this is the output of the serial monitor

[image]

logm1lo, 2024-09-28:
the fix here is to wait for a couple of hours and turn it back again. Works like a charm

closed 2024-09-28 by logm1lo

Dev-Khrish, 2024-12-18:
how did you get it to work ?

logm1lo, 2024-12-20:
> how did you get it to work ?

I still doesn't understand why I got it to work

Dev-Khrish, 2024-12-20:
Can you share a bit of schematics/screenshots of wiring or the code you used please

logm1lo, 2024-12-23:
> Can you share a bit of schematics/screenshots of wiring or the code you used please

Here it is, it's a bit messy tho:
https://github.com/cifertech/nRFBox/issues/17