cifertech/nRFBox #20: Difference between 1KOhm 0805 resistor and 1KOhm 0805 (2012X06L) resistor?

VNEvil, 2024-10-12:
Hello @cifertech  and everyone! 

In the 'nRFBOXv2 BOM.pdf' list, there are two types of resistors: 1KOhm 0805 and 1KOhm 0805 (2012X06L). Is there any difference between them? Can I use the 1KOhm 0805 (2012M) resistor to replace the 1KOhm 0805 (2012X06L) resistor?
[image]

I am looking forward to receiving a response
Thank you very much!

atsigr, 2024-10-13:
@VNEvil i think the 2012X06L. is on the metric System were is corect 2012M and 06 is the pad size as standrad all the 0805 RES are. Pretty sure you can use 1KOhm 0805 (2012M)

UrbanTrip96, 2025-02-13:
> @VNEvil i think the 2012X06L. is on the metric System were is corect 2012M and 06 is the pad size as standrad all the 0805 RES are. Pretty sure you can use 1KOhm 0805 (2012M) 

Are they interchangeable?