cifertech/nRFBox #21: .bin not working

sameepgod, 2024-10-14:
[image]

i have been trying to make this project decided to test it out on breadboard  but im not the output

mahmoudgamalmmg8, 2024-10-19:
i have same problem did u find a solution?

abasith-ak, 2024-11-04:
same bro

amir8el2, 2024-11-27:
I have this problem too. Has anyone been able to fix this problem?

ahmadjutt611, 2024-12-13:
[image]
 I have also same issue the bin is not showing any display but if upload code with ARDUINO IDE then there are many errors in code and after error resolve the display is working also analyzer working but jammer not working