cifertech/nRFBox #25: nrf-box v1

ElamrawyTamer, 2024-11-07:
Hi, can you please send the schematics of the firse nrf-box? I just don't know how to connect buttons, charging controller and, etc? It's just I already built the second version and want to test the first one. Thank you

knlklabacka, 2024-11-08:
@ElamrawyTamer can you help me with version 2?  Can you share how you got it working?  Did you order a PCB?  I tried uploading the file to JLCPCB and they came back with the board had errors.

ElamrawyTamer, 2024-11-08:
@knlklabacka i used breadboard and i did soldering with the jumper wires, because I didn't want to risk buying wrong PCBs. I also found a video on tiktok, that helped me a lot: https://www.tiktok.com/@testando_tecnologias/video/7425052647775849734

knlklabacka, 2024-11-08:
Thank you @ElamrawyTamer I will take a look.

ElamrawyTamer, 2024-11-09:
@cifertech ?

SoLongSidekick, 2024-11-30:
> @knlklabacka i used breadboard and i did soldering with the jumper wires, because I didn't want to risk buying wrong PCBs. I also found a video on tiktok, that helped me a lot: https://www.tiktok.com/@testando_tecnologias/video/7425052647775849734

No way, no freaking way. Get the fuck out of here, nice find my dude. Crap, this means I finally have to just sack up and add Tiktok to my media appetite. Seriously, thanks for posting. 

I'm... ... eh... 60-80% positive I downloaded the PCB Gerbers before v2 ever came out. I bookmarked this page well before v2 so it's very plausible. I'll dig around a bit and see if that is indeed the case, and if so I'll report back here.

AntonioDVZ, 2025-04-27:
> Hi, can you please send the schematics of the firse nrf-box? I just don't know how to connect buttons, charging controller and, etc? It's just I already built the second version and want to test the first one. Thank you

Hi. If you received the schematic version 1, can you send it to me?