cifertech/nRFBox #26: missing cpl file?

Spacedreamer0, 2024-11-13:
hello i wanted to follow the project but i don't have the required experience or tools to assemble it and wanted to have it assembled on jlcpcb but it asks me for a cpl file which i cant find. Do i miss something?

zarni99, 2024-11-28:
I also looking for cpl file and does anyone has share it here, please

nechry, 2024-12-18:
I'm also looking for the BOM and CPL for jlcpcb format.

gel3k, 2024-12-27:
also interested

knlklabacka, 2024-12-27:
Also interested

nechry, 2025-01-02:
I tried to order only PCB via jlcpcb, but the provided Gerber didn't pass the audit. is the reason:
There are minimum hole size of 0.25 mm, with a diameter of 0.35mm in your file, to ensure the quality for your order, the 4-Wire Kelvin Test and via filling are necessary, so you may need to pay for an extra cost for it.