cifertech/nRFBox #32: when a press enter in different options freezes and need to restart

SSnoody, 2024-12-24:
idk if is the new verision or my firmware is broke, i try with 2 diff esp32 and still the same thing, i actually flash both with arduino ide, the .bin way it doesnt flash the esp32

akftec, 2025-01-12:
i have same problem. did u found a solution?

akftec, 2025-01-12:
hi i found my problem. i think u have same problem 
[image]

SSnoody, 2025-01-12:
> hi i found my problem. i think u have same problem just follow this ![image](https://private-user-images.githubusercontent.com/194712128/402380723-d080e17c-ad8f-41c1-ad5e-42cc400ca090.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MDk0NzksIm5iZiI6MTczNjcwOTE3OSwicGF0aCI6Ii8xOTQ3MTIxMjgvNDAyMzgwNzIzLWQwODBlMTdjLWFkOGYtNDFjMS1hZDVlLTQyY2M0MDBjYTA5MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMlQxOTEyNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZTUyZDVkMGMxZWUyYTAzY2I0YjA0YmRkNzE1NjFkNTYxMjYzMWNjZDQxN2FhODk0MTNiNjU5NTlkYzU0YTNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iC8rlFn1VjTjA4YCn0rUuS4Nm6G4Jks_5te3XengHCY)

I have that config too, but still frozen, for example if i press the jammer it works, if i go back and press the jammer again it froze it, i cant press the same function again because it froze

akftec, 2025-01-13:
https://randomnerdtutorials.com/esp32-erase-flash-memory/ hi can u try this  if u cant acces python in terminal  try the install in microsoft store.  and i used the esp32 library version is 1.0.5 try this version after all this upload esp32  on arduino ide