cifertech/nRFBox #34: error: cannot convert 'std::string'

ahmadjutt611, 2024-12-29:
I have these error in spoofer.cpp in the switches 

switch (deviceType) {
  case 1:
    oAdvertisementData.setManufacturerData(String((char *)dataAirpods).substring(0, 31));
    break;
  case 2:
    oAdvertisementData.setManufacturerData(String((char *)dataAirpodsPro).substring(0, 31));
    break;
  case 3:
    oAdvertisementData.setManufacturerData(String((char *)dataAirpodsMax).substring(0, 31));
    break;
  case 4:
    oAdvertisementData.setManufacturerData(String((char *)dataAirpodsGen2).substring(0, 31));
    break;

wajiho23, 2024-12-29:
I have the same error: `C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\sourapple.cpp: In function 'BLEAdvertisementData getOAdvertisementData()':
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\sourapple.cpp:78:34: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   advertisementData.addData(std::string((char *)packet, 17));
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                  |
      |                                  std::string {aka std::__cxx11::basic_string<char>}
In file included from C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEServer.h:22,
                 from C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEDevice.h:21,
                 from C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\sourapple.h:9,
                 from C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\sourapple.cpp:7:
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:36:23: note:   initializing argument 1 of 'void BLEAdvertisementData::addData(String)'
   36 |   void addData(String data);  // Add data to the payload.
      |                ~~~~~~~^~~~
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp: In function 'void setAdvertisingData()':
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:197:49: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  197 |     oAdvertisementData.setManufacturerData(std::string((char *)dataAirpods, 31));
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 std::string {aka std::__cxx11::basic_string<char>}
In file included from C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEServer.h:22,
                 from C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEDevice.h:21,
                 from C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.h:9,
                 from C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:7:
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:31:35: note:   initializing argument 1 of 'void BLEAdvertisementData::setManufacturerData(String)'
   31 |   void setManufacturerData(String data);
      |                            ~~~~~~~^~~~
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:200:49: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  200 |     oAdvertisementData.setManufacturerData(std::string((char *)dataAirpodsPro, 31));
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 std::string {aka std::__cxx11::basic_string<char>}
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:31:35: note:   initializing argument 1 of 'void BLEAdvertisementData::setManufacturerData(String)'
   31 |   void setManufacturerData(String data);
      |                            ~~~~~~~^~~~
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:203:49: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  203 |     oAdvertisementData.setManufacturerData(std::string((char *)dataAirpodsMax, 31));
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 std::string {aka std::__cxx11::basic_string<char>}
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:31:35: note:   initializing argument 1 of 'void BLEAdvertisementData::setManufacturerData(String)'
   31 |   void setManufacturerData(String data);
      |                            ~~~~~~~^~~~
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:206:49: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  206 |     oAdvertisementData.setManufacturerData(std::string((char *)dataAirpodsGen2, 31));
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 std::string {aka std::__cxx11::basic_string<char>}
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:31:35: note:   initializing argument 1 of 'void BLEAdvertisementData::setManufacturerData(String)'
   31 |   void setManufacturerData(String data);
      |                            ~~~~~~~^~~~
C:\Users\wajih\Downloads\nRFBox-2.2.1\nRFBox-2.2.1\nRFBox_V2\spoofer.cpp:209:49: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  209 |     oAdvertisementData.setManufacturerData(std::string((char *)dataAirpodsGen3, 31));
[... cut ...]
      |                                                 std::string {aka std::__cxx11::basic_string<char>}
C:\Users\wajih\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\BLE\src/BLEAdvertising.h:31:35: note:   initializing argument 1 of 'void BLEAdvertisementData::setManufacturerData(String)'
   31 |   void setManufacturerData(String data);
      |                            ~~~~~~~^~~~

exit status 1

Compilation error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'`

cifertech, 2024-12-29:
Hi, use version 1.0.5 of the ESP32 board and follow the instructions.

https://github.com/cifertech/nRFBox?tab=readme-ov-file#-uploading-the-firmware-bin-to-esp32

ahmadjutt611, 2024-12-30:
can you please give us the library's name and versions like BLE

ahmadjutt611, 2024-12-30:
[image]

after downgrading the esp version. Now shown this error.

ahmadjutt611, 2024-12-30:
If we install .bin then there is no display on the OLED

closed 2024-12-30 by ahmadjutt611