cifertech/nRFBox #40: OLED Display Not Working After Program Upload

MohammadNabin, 2025-01-20:
After successfully compiling and uploading the nRFBox firmware to an ESP32 with an I2C-connected OLED display, the display remains black and does not show any output. The compilation errors related to exceeding program storage space were resolved by changing the partition scheme, but the OLED issue persists.

Steps to Reproduce:
Clone the latest nRFBox firmware repository.
Configure the project for an ESP32 board and ensure the correct wiring for an I2C OLED (e.g., SSD1306 128x64).
Adjust the partition scheme to "Minimal SPIFFS (1.9MB APP, 190KB SPIFFS)" in Arduino IDE.
Upload the program to the ESP32.
Observe that the OLED display remains black.

Environment Details
Board: ESP32 Dev Module
OLED Display: SSD1306 128x64 (I2C interface)
Arduino IDE Version: 2.3.4
Partition Scheme: Minimal SPIFFS (1.9MB APP, 190KB SPIFFS)
Firmware Version: v2.5.1
Libraries:
U8g2 for OLED display handling
Other dependencies from the repository
Troubleshooting Steps Taken
Verified wiring (VCC, GND, SCL, SDA) to ensure proper connections.
Checked the I2C address using an I2C scanner:
Result: The OLED was detected at 0x3C.
Tested the display with a minimal U8g2 example sketch:
Result: The display worked as expected with the example code.

DotNetInAction, 2025-01-26:
Same issue with ESP32 Wroom 32U. Code compiles and uploads but nothing on the screen. Tried taking things out line by line by removing neopixel, eeprom, radios but still nothing on the screen.

Flashing any previous version works straight away. Flashing latest bin getting same issue with blank screen.

RoToSpLaSh, 2025-01-29:
Make sure the SCL pin 22 and  SDA pin 21

dev-shrubb3ry, 2025-01-29:
Im pretty sure it is not a screen issue. I tried a minimal sketch with only usg8 library and screen does work. I have also tried another esp32 with the integrated screen and flashing previous image works straight away. Flashing newest nothing comes up. Sometimes I do observe what it seems a bootloop in a serial monitor tho.

It is a shame debugging is so hard on esp32 and arduino ide. I feel like it is most likely a dependency failure which would come up straight away. I been comparing previous tag code with newest and EEPROM looks like a good candidate as it gets initialized before anything comes up on screen and it was added in latest version.

dev-shrubb3ry, 2025-01-29:
Ok, I got that working.

It appeared to be esp32 issue. Switching to different board worked straight away. Flashing latest .bin still sends it into a bootloop, but compiling code and uploading directly works just fine.

marikuns, 2025-01-31:
Look spi address of your display and use `u8g2.setI2CAddress(0x78);` before begin. replace 0x78 with your value