cifertech/nRFBox #49: Capacitors C8/C9 no voltage specified

GreyHatUser, 2025-03-19:
I built the entire thing, i noticed first time i ordered the capacitors wrong as i bought 10uf 10v capacitors, when in the schematic specifies 2.2uf, now, when i powered the device on both (of the wrong capacitors) got burnt, i wanted to make a **new order for the right capacitors for C8/C9**, but there is **no voltage specified**, **does anyone know what voltage should they be**?

Still, trying to connect the device with no C8/C9 capacitors wich should theoretically work, the tool doesnt get detected by the pc, i changed the USB  to UART chip in case it was broken but still nothing. Are the C8/C9 100% necessary?

UrbanTrip96, 2025-03-20:
[image]

В gerber файлах указано C8/C9 Tant 2.2uf

Я временно заменил на керамические конденсаторы 2.2uf и вроде всё работает. Компьютер видет esp32 и даже записывает прошивку, но не включается. Возможно дело в прошивке, пока не разобрался.

SunboX, 2025-04-22:
@UrbanTrip96 Do you have the complete Altium project? If so, could you send it to me please? I would like to do some changes.