cifertech/nRFBox #53: bin 2.5.0 not working. Blank screen, no output

Believeinus, 2025-04-02:
bin 2.5.0 showing black screen on esp32 30 pin. 2.2.1 works but 2.5.0 is black screen, no output

Nicoldik, 2025-04-11:
hi, I have this problem too, I suspect it's either esp32 or oled screen.

Nicoldik, 2025-04-13:
hello, I understand what the problem is. the problem is that esp32 is limited in 4MB versions (let's not mention versions with 8mb and 16mb of memory) of files up to 1.31 and the firmware weighs 1.54MB. you need (the firmware author) to make bootloader.bin and partitions.bin for the firmware

kosmostrater, 2025-04-28:
Same issue, 2.2.1 flashed ok and displays on the oled, but 2.5.0 does not display.

Nicoldik, 2025-05-20:
[kosmostrater](https://github.com/kosmostrater) the line "E (505) esp_image: Image length 1553536 doesn't fit in partition length 1310720
E (505) boot: OTA app partition slot 0 is not bootable
E (505) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (513) boot: OTA app partition slot 1 is not bootable
E (518) boot: No bootable app partitions in the partition table" says that the file .bin is heavy, so try using the .ino file.