cifertech/nRFBox #64: OLED display doesn’t work (ESP32 C6)

Andrew32151, 2025-05-09:
Hello

I tried for the first time to do this kind of circuits like Nrf box v2 and Uploaded the filmware on ESP32 C6 and after restarting it the Oled screen i connected to it doesnt shows up something 

Can somone help to solve this problem ??

RoToSpLaSh, 2025-05-10:
Do you try change pin SDL and SCA?

Andrew32151, 2025-05-10:
> Do you try change pin SDL and SCA?

No i have SDL=21 SCA=22 so were to try to put them now ?

(Also im not sure but i think after a little research that it will work only on 0.96 oled screen because it's SSD1306 and my oled 1.3 that has SH1106)

RoToSpLaSh, 2025-05-10:
> > Do you try change pin SDL and SCA?
> 
> No i have SDL=21 SCA=22 so were to try to put them now ?
> 
> (Also im not sure but i think after a little research i found that it will work only on a 0.96 oled screen because it's SSD1306 and not on my oled 1.3 that has SH1106)

Yes ,only on 0.96 oled, with ssd1306, I didn't pay attention to the screen model you had specifically, sorry!!

Andrew32151, 2025-05-10:
Ohh ok nevermind, so now I'm trying to install bluejammer and now it sent me an error and tell me to reset the ESP32 or PRESS BOOT but nothing works so I guess I go to blue jammer forum for this.

also thank you so much for your time to help me out!!