cifertech/nRFBox #66: Charging Battery released Magic Smoke

SunboX, 2025-06-17:
I did build a nRFBOX V2 which is working fine so far.

But when I did try to charge a battery, it worked fine at first but after some minutes it released the "Magic Smoked". Can someone tell what I did wrong?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stonedDiscord, 2025-06-18:
The capacitors C8 and C9 are installed in reverse.

[image]

stonedDiscord, 2025-06-18:
The markings on the PCB indicate the negative side, the line on the capacitor itself marks the positive side.

gel3k, 2025-06-25:
hey, where did you get this 'battery holder' pcb ?

SunboX, 2025-06-25:
@gel3k I made it myself. ;) With KiCAD it's very easy.